chaeplin/esp8266_and_arduino #32: HX711 - ap_setting.h no such file or directory

kozoke, 2016-05-29:
Hi,
I'm trying to compile the 03-hx711-scale-esp-12.ino in the arduino ide 1.6.9.  I'm getting the following error. I've added the folder structure in the arduino folder and i've added it to my libraries file but no luck.  any other suggestions?  I'm trying to run it on an esp 8266 12e.
Thanks!

Error:

hx711scaleesp12k.ino:17:39: fatal error: /usr/local/src/ap_setting.h: No such file or directory

 #include "/usr/local/src/ap_setting.h"

chaeplin, 2016-05-29:
do like this. ap_setting.h has dome define.
```
//#include "/usr/local/src/ap_setting.h"
// ****************
#define WIFI_SSID "WIFI_SSID_CHAGE_THIS"
#define WIFI_PASSWORD "WIFI_PASS_CHAGE_THIS"
// change below
#define MQTT_SERVER { 192, 168, 10, 10 }
//
#define OTA_PASSWORD "OTA_PASSWORD_CHANGE_THIS"
```

kozoke, 2016-05-29:
Hi, Thank you for your quick response. I modified the file (appreciate the example) which addressed the ap_settings.h issue.  Now I am getting a series of effort around the average.h library.

If it helps, i'm simply trying to use a piece of code to pull values from the load cell.  That's all.  I've tried numerous libraries/scripts but none of them seem to work on the esp826612e. I even saw on Reddit your contribution on someone's hx711 issue - that script did not work for me either. error is < ets Jan 8 2013,rst cause:2, boot mode:(3,6)  >

Appreciate your help and any additional thoughts.

Error messages:
In file included from C:\Users\kkozo\Documents\Arduino\esp8266_and_arduino-master\_03-hx711-scale\_03-hx711-scale-esp-12\_03-hx711-scale-esp-12.ino:9:0:

C:\Users\kkozo\Documents\Arduino\libraries\ESP8266Ping/ESP8266Ping.h:26:5: error: stray '\302' in program

     <title>ESP8266Ping/ESP8266Ping.h at master Â· dancol90/ESP8266Ping</title>

     ^

C:\Users\kkozo\Documents\Arduino\libraries\ESP8266Ping/ESP8266Ping.h:26:5: error: stray '\267' in program

In file included from C:\Users\kkozo\Documents\Arduino\esp8266_and_arduino-master\_03-hx711-scale\_03-hx711-scale-esp-12\_03-hx711-scale-esp-12.ino:9:0:

C:\Users\kkozo\Documents\Arduino\libraries\ESP8266Ping/ESP8266Ping.h:114:3: error: missing terminating ' character

   <form accept-charset="UTF-8" action="/dancol90/ESP8266Ping/search" class="js-site-search-form" data-scoped-search-url="/dancol90/ESP8266Ping/search" data-unscoped-search-url="/search" method="get"><div style="margin:0;padding:0;display:inline"><input name="utf8" type="hidden" value="&#x2713;" /></div>

   ^

In file included from C:\Users\kkozo\Documents\Arduino\esp8266_and_arduino-master\_03-hx711-scale\_03-hx711-scale-esp-12\_03-hx711-scale-esp-12.ino:9:0:

C:\Users\kkozo\Documents\Arduino\libraries\ESP8266Ping/ESP8266Ping.h:234:11: error: missing terminating ' character

           <form accept-charset="UTF-8" action="/logout" class="logout-form" data-form-nonce="310d3d86c1c00d4aabf3b393d5d3c91e252a56b7" method="post"><div style="margin:0;padding:0;display:inline"><input name="utf8" type="hidden" value="&#x2713;" /><input name="authenticity_token" type="hidden" value="s+bYWXfzau0xno6vU6iwjXw10NrlQ/3NsdI2/CpZsoib8OrdR63GTOw4L2fIilbqhmHAT+9GMUa2WcPNmEW8bA==" /></div>

           ^

C:\Users\kkozo\Documents\Arduino\libraries\ESP8266Ping/ESP8266Ping.h:271:9: error: missing terminating ' character

         <form accept-charset="UTF-8" action="/notifications/subscribe" class="js-social-container" data-autosubmit="true" data-form-nonce="310d3d86c1c00d4aabf3b393d5d3c91e252a56b7" data-remote="true" method="post"><div style="margin:0;padding:0;display:inline"><input name="utf8" type="hidden" value="&#x2713;" /><input name="authenticity_token" type="hidden" value="X+HGFWypzPuEIXVy0hGPEgNw2B/ccq9KnJGIGveVUEdlHYXhCi4ESNUGf/UxV+9Qpoet1pmUuSLpuZllB5DOrA==" /></div>      <input class="form-control" id="repository_id" name="repository_id" type="hidden" value="39532736" />

         ^

In file included from C:\Users\kkozo\Documents\Arduino\esp8266_and_arduino-master\_03-hx711-scale\_03-hx711-scale-esp-12\_03-hx711-scale-esp-12.ino:9:0:

C:\Users\kkozo\Documents\Arduino\libraries\ESP8266Ping/ESP8266Ping.h:300:81: error: stray '@' in program

                     <span class="description">Be notified when participating or @mentioned.</span>

                                                                                 ^

C:\Users\kkozo\Documents\Arduino\libraries\ESP8266Ping/ESP8266Ping.h:346:5: error: missing terminating ' character

     <form accept-charset="UTF-8" action="/dancol90/ESP8266Ping/unstar" class="starred" data-form-nonce="310d3d86c1c00d4aabf3b393d5d3c91e252a56b7" data-remote="true" method="post"><div style="margin:0;padding:0;display:inline"><input name="utf8" type="hidden" value="&#x2713;" /><input name="authenticity_token" type="hidden" value="ULhdKcl3nj6i/8JWwZY/1wL8kaLVRHAYgeppJNEJiyRt8igKuhrEGKM2SaY0nhUkpg5lkpqdA5HskILaHhsBPg==" /></div>

     ^

C:\Users\kkozo\Documents\Arduino\libraries\ESP8266Ping/ESP8266Ping.h:358:5: error: missing terminating ' character

     <form accept-charset="UTF-8" action="/dancol90/ESP8266Ping/star" class="unstarred" data-form-nonce="310d3d86c1c00d4aabf3b393d5d3c91e252a56b7" data-remote="true" method="post"><div style="margin:0;padding:0;display:inline"><input name="utf8" type="hidden" value="&#x2713;" /><input name="authenticity_token" type="hidden" value="q62oh76lO4X10B7lIvZrvAzvghmvdwvPzQ+jM1od4bnapzsood7j60GudbfKxnP4VmZgm811gUZez712GzHG6w==" /></div>

     ^

C:\Users\kkozo\Documents\Arduino\libraries\ESP8266Ping/ESP8266Ping.h:374:11: error: missing terminating ' character

           <form accept-charset="UTF-8" action="/dancol90/ESP8266Ping/fork" class="btn-with-count" data-form-nonce="310d3d86c1c00d4aabf3b393d5d3c91e252a56b7" method="post"><div style="margin:0;padding:0;display:inline"><input name="utf8" type="hidden" value="&#x2713;" /><input name="authenticity_token" type="hidden" value="UuOa+GZbxyE4q4hctnXW1Gopq2SN6fDYAHe5JqxzKGClZvlE3SeVLZgj1QeA0a33BKS6dfwzye8qeIXtGW39EQ==" /></div>

           ^

In file included from C:\Users\kkozo\Documents\Arduino\esp8266_and_arduino-master\_03-hx711-scale\_03-hx711-scale-esp-12\_03-hx711-scale-esp-12.ino:9:0:

C:\Users\kkozo\Documents\Arduino\libraries\ESP8266Ping/ESP8266Ping.h:564:9: error: missing terminating ' character

[... cut ...]
                                                                                                                                                                                         ^

C:\Users\kkozo\Documents\Arduino\libraries\Average/Average.h:973:199: error: expected constructor, destructor, or type conversion before '(' token

         <td id="LC85" class="blob-code blob-code-inner js-file-line"><span class="pl-k">template </span>&lt;<span class="pl-k">class</span> <span class="pl-en">T</span>&gt; Average&lt;T&gt;::Average(<span class="pl-c1">uint32_t</span> size) {</td>

chaeplin, 2016-05-30:
Use this one : https://github.com/chaeplin/esp8266_and_arduino/tree/master/_03-hx711-scale/old_scale_version/_03-hx711-scale-v1-adaptor

kozoke, 2016-05-30:
Hi,
Thanks. I had looked at this sketch before but had similar problems. But, I
tried it again, configured ap_setting.h and deleted and redownloaded the
pubsubclient.  No luck. Here are the errors.

Any other suggestions?
Is there a segment of this code that just deals with the hx711? I use
HomieOTA for my mqtt/wifi connectivity so I really just need to figure out
how to read from the HX711.

appreciate your help.

_03-hx711-scale-esp-12:25: error: no matching function for call to
'PubSubClient::PubSubClient(WiFiClient&, IPAddress&)'

 PubSubClient client(wifiClient, server);

                                       ^

C:\Users\kkozo\Documents\Arduino\esp8266_and_arduino-master\_03-hx711-scale\old_scale_version\_03-hx711-scale-v1-adaptor\_03-hx711-scale-esp-12\_03-hx711-scale-esp-12.ino:25:39:
note: candidates are:

In file included from
C:\Users\kkozo\Documents\Arduino\esp8266_and_arduino-master\_03-hx711-scale\old_scale_version\_03-hx711-scale-v1-adaptor\_03-hx711-scale-esp-12\_03-hx711-scale-esp-12.ino:1:0:

C:\Users\kkozo\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:116:4:
note: PubSubClient::PubSubClient(const char*, uint16_t,
std::function<void(char*, unsigned char*, unsigned int)>, Client&, Stream&)

    PubSubClient(const char*, uint16_t, MQTT_CALLBACK_SIGNATURE,Client&
client, Stream&);

    ^

C:\Users\kkozo\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:116:4:
note:   candidate expects 5 arguments, 2 provided

C:\Users\kkozo\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:115:4:
note: PubSubClient::PubSubClient(const char*, uint16_t,
std::function<void(char*, unsigned char*, unsigned int)>, Client&)

    PubSubClient(const char*, uint16_t, MQTT_CALLBACK_SIGNATURE,Client&
client);

    ^

C:\Users\kkozo\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:115:4:
note:   candidate expects 4 arguments, 2 provided

C:\Users\kkozo\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:114:4:
note: PubSubClient::PubSubClient(const char*, uint16_t, Client&, Stream&)

    PubSubClient(const char*, uint16_t, Client& client, Stream&);

    ^

C:\Users\kkozo\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:114:4:
note:   candidate expects 4 arguments, 2 provided

C:\Users\kkozo\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:113:4:
note: PubSubClient::PubSubClient(const char*, uint16_t, Client&)

    PubSubClient(const char*, uint16_t, Client& client);

    ^

C:\Users\kkozo\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:113:4:
note:   candidate expects 3 arguments, 2 provided

C:\Users\kkozo\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:112:4:
[... 153 lines cut ...]
                                  ^

_03-hx711-scale-esp-12:30: error: expected ')' before '&' token

_03-hx711-scale-esp-12:30: error: expected initializer before 'pub'

 void callback(const MQTT::Publish& pub) {

                                    ^

C:\Users\kkozo\Documents\Arduino\esp8266_and_arduino-master\_03-hx711-scale\old_scale_version\_03-hx711-scale-v1-adaptor\_03-hx711-scale-esp-12\_03-hx711-scale-esp-12.ino:
In function 'void setup()':

_03-hx711-scale-esp-12:48: error: 'class PubSubClient' has no member named
'set_callback'

   client.set_callback(callback);

          ^

_03-hx711-scale-esp-12:48: error: 'callback' was not declared in this scope

   client.set_callback(callback);

                       ^

chaeplin, 2016-05-30:
hmm check this https://gist.github.com/chaeplin/d3cc3675982d8c92455b
It's https://github.com/bogde/HX711 example with ESP.wdtDisable.

kozoke, 2016-05-30:
FYI I took your code - prior suggestion _03 and pulled out all of the
wifi/mqtt stuff.  Good news is that it is not doing the soft reset. Bad
news, it is only giving me a raw value of 65535.  I had the hardware (hx711
and load sensors working correctly with an arduino yesterday so they should
be fine.  Here is what was left of your code when I stripped it down.
Anything in here look wrong?  I'll try your suggestion above now.  Thanks!

#include <HX711.h>

#if defined(ESP8266)
#include <pgmspace.h>
#else
#include <avr/pgmspace.h>
#endif
#include <Wire.h>  /

// wifi
#ifdef __IS_MY_HOME
  #include "/usr/local/src/ap_setting.h"
#else
  #include "ap_setting.h"
#endif

int inuse = LOW;
#define nemoisOnPin 13

void setup() {
  Serial.begin(38400);
  //Wire.pins(4, 5);
  Wire.begin(4,5);
  Serial.println("HX711 START");
  delay(20);
  pinMode(nemoisOnPin,INPUT);
}

void loop() {
  inuse = digitalRead(nemoisOnPin);
  if ( inuse == HIGH ) {
    requestHx711();
  }
  delay(500);
}

String macToStr(const uint8_t* mac)
{
  String result;
  for (int i = 0; i < 6; ++i) {
    result += String(mac[i], 16);
    if (i < 5)
      result += ':';
  }
  return result;
}

void requestHx711() {
     Wire.requestFrom(2, 2);

     int x;
     byte a, b;

     a = Wire.read();
     b = Wire.read();

     x = a;
     x = x << 8 | b;

    Serial.print(x);

     String payload = "{\"NemoWeight\":";
     payload += x;
     payload += "}";

    sendHx711(payload);
}

chaeplin, 2016-05-30:
This sketch is not esp8266 + hx711.
It's esp8266 + pro mini(+ hx711).

So Arduino should have i2c connection to esp8266.

https://github.com/chaeplin/esp8266_and_arduino/blob/master/_03-hx711-scale/pics/6.jpg

kozoke, 2016-05-30:
Hi,
Just tried your suggestion above: It's https://github.com/bogde/HX711 example
with ESP.wdtDisable.
That didn't work - took me back to the bootloader message: ets Jan 8
2013,rst cause:2, boot mode: (3,6)

Seems like your _03 older script works (at least it isn't showing the
bootloader message).
I just confirmed the hardware is working with an arduino script called
"simple_scale" (forget where i got it).  I tried putting the Simple_Scale
script (below) on the esp, changing the pings to d4,d5 and I got a fatal
exception messages spooling across the screen.

Well, we are getting closer!  any other ideas?
thanks

kozoke, 2016-05-30:
I'm not sure i'm following.
I pulled your code as suggested: esp8266_and_arduino
<https://github.com/chaeplin/esp8266_and_arduino>/_03-hx711-scale
<https://github.com/chaeplin/esp8266_and_arduino/tree/master/_03-hx711-scale>
/old_scale_version
<https://github.com/chaeplin/esp8266_and_arduino/tree/master/_03-hx711-scale/old_scale_version>
/_03-hx711-scale-v1-adaptor
<https://github.com/chaeplin/esp8266_and_arduino/tree/master/_03-hx711-scale/old_scale_version/_03-hx711-scale-v1-adaptor>
/_03-hx711-scale-esp-12
<https://github.com/chaeplin/esp8266_and_arduino/tree/master/_03-hx711-scale/old_scale_version/_03-hx711-scale-v1-adaptor/_03-hx711-scale-esp-12>
/_03-hx711-scale-esp-12.ino

I pulled out the wifi and mqtt stuff and it ran but with only one repeating
value.

You mention arduino hsould have i2c which I believe is included wire.h
library which is still in the remaining code I sent.  Also, please note i'm
using a esp8266 12e dev board if that matters.

Clearly i'm missing something here.  Am I using the old or wrong wire.h
library?

thx

chaeplin, 2016-05-30:
Which esp do ypu use?

kozoke, 2016-05-30:
Esp8266 12e amica

Keith Kozo
415 937 1909

chaeplin, 2016-05-30:
Esp8266 use 3.3V.
Which arduino board do you use ?

If arduino board use 5V, you need i2c level shifter.
http://playground.arduino.cc/Main/I2CBi-directionalLevelShifter

kozoke, 2016-05-30:
Hi
Arduino uno. 

I have the esp powered through the usb 5v pin. I'll look at the guidance about level shifting. That is new to me

Thx

chaeplin, 2016-05-30:
Uno i2c use 5V, esp i2c use 3.3V.
So voltage level shifting is needed.
https://www.youtube.com/watch?v=VDfI546YThQ

kozoke, 2016-05-30:
Ok. Let me look into it and get back to you. Again, thanks for your patience and support.

chaeplin, 2016-05-30:
Check this for i2c.
http://gammon.com.au/i2c
https://learn.sparkfun.com/tutorials/i2c

kozoke, 2016-05-30:
Ok but...  The hx711 takes 2.6 to 5v input. If I just change input voltage to the 3.3v pin instead of the 5v pin wouldn't that work?  Or am I missing something?

chaeplin, 2016-05-30:
@kozoke 
Which combination do you use ? Let me check again.
1) esp8266 + hx711
2) esp8266 --i2c--arduino-hx711

kozoke, 2016-05-30:
esp8266 + hx711

-- 
Keith Kozo
4159371909
krk112@alumni.psu.edu

chaeplin, 2016-05-30:
hmm 1) then you should check https://gist.github.com/chaeplin/d3cc3675982d8c92455b
It should work. If not, change pins sick, out.
When I use more than 5 in scale.read_average(?) and scale.get_value(?) error occurred.
After uploading sketch, reset once more for sure.

kozoke, 2016-05-30:
Tried the sketch above. No luck. I'm back to the "ets Jan 8 2013,rst cause:2, boot mode: (3,6)"  Same error if I use the 3.3v pin or the 5v pin.

I did not understand your statement.  "If not, change pins sick and dout - you mean change the sda and clock pins?  they are on d4 for data and d5 for clock.  Those are the defaults I believe?

You also said, "When I use more than 5 in scale.read_average(?) and scale.get_value(?) error occurred."  Do you mean, when you use 5v vs 3.3v you see the monitor? Don't understand the reference to scale.read_average and scale.get_value.  Can you clarify?

I also tried re-downloading the bodge/hx711 library you refer to in your readme fiile.  That didn't help and made the other script that kinda worked (showed values) now breaks with the ets error message. The bodge library is different that the one I was using. 

I'm just baffled by this. Do you have the esp12 and hx711 or is your set up different?

chaeplin, 2016-05-31:
1) HX711(byte dout, byte pd_sck, byte gain = 128);
   --> out to 4, sick to 5(4, 5 is not fixed, can use other pins)
2) esp vcc(3.3) to vcc of hx711, and esp gnd to gnd of hx711
3) looks like download err to me. How do you download hx711 library ?

kozoke, 2016-05-31:
1) OK yes, understand that.  I'll double check the pins. Assume 4 =D4 on my
board and =D5 on my board (esp 8266 12e).  That makes sense.
2) Yes - agree on 3.3 to vcc and ground to ground. all good there.
3) I went to the link and downloaded the zip from the site you refer to -
https://github.com/bogde/HX711
I then go into board manager in the arduino IDE and import it.  Then I go
in and ensure the .h file is in the HX711 directory (not in a second sub
directory).

I'm sure i'm missing something obvious... Just don't know what it is, yet.

I think i'm going to set up another pc with an arduino IDE and reinstall
just your recommended ino and libarary (with nothing else) and give that a
try.

I also posted another message on the www.8266.com site to see if anyone
else has had any luck with this config - esp826612e -
http://www.amazon.com/HiLetgo-Version-NodeMCU-Internet-Development/dp/B010O1G1ES?ie=UTF8&psc=1&redirect=true&ref_=oh_aui_detailpage_o02_s01

Linked directly to:
hx711
(
http://www.amazon.com/SMAKN%C2%AE-Weighing-Dual-channel-Conversion-Shieding/dp/B00M6POINC?ie=UTF8&psc=1&redirect=true&ref_=oh_aui_detailpage_o01_s00
)

chaeplin, 2016-05-31:
download https://github.com/bogde/HX711 and unzip it.
Find Arduino library folder, move unzipped lib to hat folder.

https://www.youtube.com/watch?v=tcpE19KaKis
https://www.youtube.com/watch?v=Zdxz3YUd1S8

kozoke, 2016-06-01:
Solved.  I went through and blew away all remants of anything hx711 in the libraries/scripts. Downloaded the Bogde and tried that. I also recall something about pin assignments so I added this to the code: (and changed the .ino pins to 4 for data, 5 for clock)
*****************************************************************
#ifndef Pins_Arduino_h
#define Pins_Arduino_h

#include "../generic/common.h"

static const uint8_t SDA = 4;
static const uint8_t SCL = 5;

static const uint8_t LED_BUILTIN = 16;
static const uint8_t BUILTIN_LED = 16;

static const uint8_t D0   = 16;
static const uint8_t D1   = 5;
static const uint8_t D2   = 4;
static const uint8_t D3   = 0;
static const uint8_t D4   = 2;
static const uint8_t D5   = 14;
static const uint8_t D6   = 12;
static const uint8_t D7   = 13;
static const uint8_t D8   = 15;
static const uint8_t D9   = 3;
static const uint8_t D10  = 1;

#endif /* Pins_Arduino_h */
*******************************************************************
And, it kinda worked but was getting a reading but also a stack error.  

So, I tried this one (simple scale - which seemed to work before).
https://github.com/queuetue/Q2-HX711-Arduino-Library
And it worked!  Responded well to changes in load sensor pressure!!!  So, it was a library problem or more likely a pin assignment.

Thanks again for your help patience and support. I hope this helps someone else in the future.

chaeplin, 2016-06-01:
@kozoke Oh great. Thanks for info on new lib.

closed 2016-06-01 by chaeplin

broskunta, 2017-02-18:
@kozoke Please can I see a sample code that worked. I am having issues setting up the ESP8266 to work with the HX711 library.
Thanks.

kozoke, 2017-02-21:
Hi,
This is my code for a homie (wifi/mqtt) dog food bowl scale.  It isn't very
accurate but good enough to flag when my dog gets food and is done (weight
rises and drops below a certain point).

That said, the code works.
Does that help?