chaeplin/esp8266_and_arduino #38: Doubts about Deep Sleep and Light Sleep

arlaor, 2016-12-09:
Hello @chaeplin 
To put my doubts in context, I tell you quickly what I am doing:
My project consists of a module (1) ESP-12E connected to an electrical corner, which allows me to open the door via my telephone or via a button that works as a remote control via WIFI with a client module (2) ESP-12E Connects to (1).
The module (2) works with 3 AAA batteries total 3.8V batteries and a Momentary Push button that allows me to activate or deactivate the circuit.
The project is done with websockets [@Links2004](https://github.com/Links2004/arduinoWebSockets)
The remote control or module (2) remains in idle for undefined periods of time waiting to be used again to open the door.
I want to implement Light Sleep or Deep Sleep, to reduce unnecessary battery consumption while not in use.
As the batteries have a switching on or off to disconnect the batteries, it was planned to turn it off in the standby times, but it is not possible, since when switching it off, a proper disconnection of module (2) and module (1) , Causing the modulus (1) to lock.
My questions are as follows:
1.- In order to use light sleep or deep sleep, should I always use pins 16 and RST to reactivate the module? Or is there any way that I can activate the module with the same Momentary Push button that I use to send the door opening command to module (1)?
In my project what I do in module (2) is to connect the GPIO16 via Momentary Push button to a 3.3V pin and read the high or low status of GPIO16.

2.- When activating Deep Sleep or Light Sleep in an Esp-12E module, does it save actions that have been previously executed by this one? Example, if I have activated a Gpio and in a webpage with websockets I show interactively that this pin is activated, when it enters to deep Sleep or light Sleep and after leaving that state, it keeps updated the information with which entered mode Of deep sleep?

chaeplin, 2016-12-09:
1) deepsleep : To wakeup module, reset is needed. GPIO16 generate very short low pulse to reset module.

2) deepsleep : no syate preserved. need to save status to rtc mem or flash.

https://github.com/chaeplin/esp8266_and_arduino/blob/master/_36-esp-rtc-mem-test/_36-esp-rtc-mem-test.ino
https://github.com/chaeplin/esp8266_and_arduino/tree/master/_48-door-alarm-deepsleep

arlaor, 2016-12-12:
Hi @chaeplin 
Can I use this library for [pubsubclient](https://github.com/knolleary/pubsubclient/blob/master/src/PubSubClient.h)?
Instead of using
```javascript
#include "/usr/local/src/ap_setting.h"
```
I can use it directly
```javascript
#include "ap_setting.h"
```
for the case
```javascript
#include "ap_setting.h"
```

```javascript
#define WIFI_SSID "AP_SSID"
#define WIFI_PASSWORD "AP_PASSWORD"

#define WIFI_BSSID { 0xaa, 0xaa, 0xaa, 0xaa, 0xaa, 0xaa }
#define WIFI_CHANNEL 10
#define MQTT_SERVER { 192, 168, 10, 10 }
//
#define OTA_PASSWORD "OTA_PASSWORD"
```
Having as a reference the doubt raised in previous message, it is necessary to handle all the parameters that it contains "ap_setting.h"
This line #define MQTT_SERVER { 192, 168, 10, 10 }  refers to the IP with which the ESP-12E is connected to the router or the IP with which the router is connected to the Internet?

#define OTA_PASSWORD "OTA_PASSWORD" is required?
#define WIFI_CHANNEL 10  is required?

for the case
```javascript
#include "user_interface.h"
```
Can I use this library for [user_interface.h](https://github.com/esp8266/Arduino/blob/master/tools/sdk/include/user_interface.h)?

chaeplin, 2016-12-12:
* I use ap_setting.h to hide ssid and password of ap and mqtt, so you can use constant at sketch or file.
* I use knolleary's pubsubclient as you linked.
* MQTT_SERVER is ip of mosquitto server
* OTA_PASSWORD is used to update firmware using Arduino/IDE https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/readme.md
* WIFI_CHANNEL - no need

arlaor, 2017-01-07:
Hello @chaeplin happy New Year to you

more question
1. Is it possible to use arduinota.h to update from a web page as it happens with ESP8266HTTPUpdateServer.h or is it not possible?
Or when using the two libraries in my sketch, these work each one separately?

2. If each one works separately, do you know which line of code is in the ESP8266HTTPUpdateServer.h library that uploads the .bin file to the ESP-12E module?
I want to use this line of code to determine if I should wait to enter deepsleep mode, while updating my ESP-12E module.

3. Can you please tell me if to know the voltage that circulates in the ESP-12E module, I must use some physical connection between the pin A0 (ADC0) and some voltage pin 3.3V or I only have to enter these lines of code:
```javascript
unsigned int vbatt;
// for battery check
ADC_MODE(ADC_VCC);
vbatt = ESP.getVcc() * 0.96;
```
and my ESP- 12E internally calculate the voltage that circulates in this?

another question. In which pins should I locate the multimeter to directly measure the voltage that my ESP module consumes?

another question. I'm using an ESP-12E module as a kind of remote control to open a door. Then my module, after executing the command to open the door, my dulo enters deepsleep mode, until re-execute the command to open the door.
The question is. Is it enough to save battery, use the following lines of code:
```javascript
...
long lastbattery = 0;
const int sleepTimeS = 2;
...
    switch(type) {
        case WStype_DISCONNECTED:
            Serial.printf("[WSc] Disconnected!\n");
            //digitalWrite(LED_BUILTIN, HIGH);
            Serial.println("voy a dormir");
            ESP.deepSleep(sleepTimeS * 1000000);
            break;
        case WStype_CONNECTED:
              {
                Serial.printf("[WSc] Connected to url: %s\n",  payload);
				webSocket.sendTXT("D13");Serial.println("enviado D13");

       digitalWrite(D4, LOW);
       if(bt==0){lastbattery=millis();}
       b=1;
            }
            break;
...
void loop() {
...
 if(millis() - lastbattery >=20000 && b==1){ 
  webSocket.sendTXT("C12");
      // Close socket and wait for disconnect from WiFi
   client.stop();
    if ( WiFi.status() != WL_DISCONNECTED ) {
      WiFi.disconnect();
    }
    b=0;
 }
```
or should I use other lines of code?

another question. What code lines should I use to save battery power, sleeping my ESP-12E module and allowing me to wake it after a certain time, without using any external action, such as a button or an external reset?

arlaor, 2017-01-14:
Hello @chaeplin 
You have any idea?

closed 2021-07-15 by chaeplin