chaeplin/esp8266_and_arduino #40: Your circuit with self wake up

Fettkeewl, 2017-04-25:
Hi chaeplin, thank you for this great circuitry. 
I'm having issues making it work while also maintaining the ability to wake my device by software as well as an external event. 

By using your circuit for the door alarm combined with a direct connection between gpio16 and rst I cannot wake my device from deep sleep with the reed switch. 

I've deducted from various testings that the issues are with voltages. I can reset my device with your circuit as long as my esp is awake because the voltage at rst is held at 3.19V directly tied to gpio16. When the esp sleeps, voltage at rst is 3.3V and your circuit does not work anymore. I thick this somehow blocks rst from being grounded through the circuitry you have provided. 

I've been thinking of adding a voltage divider between my gpio16 to rst connection to mimic the voltage I measured at rst while my esp is awake. 
Do you think this is a valid solution or have I misunderstood it all? 

See my sketch:
https://www.dropbox.com/s/141j6c6skuilrls/20170425_195826.jpg?dl=0

I'd appreciate your help, what I need is simple in concept, my esp to be able to awake by itself a well as upon a door being opened(Reed switch)

chaeplin, 2017-04-25:
https://github.com/chaeplin/esp8266_and_arduino/tree/master/_48-door-alarm-deepsleep

check circuit, RST is only tied to a TR.

Fettkeewl, 2017-04-25:
Yes I know, I understand your circuit, I was wondering if you had any idea as how to modify it to be able to wake esp by software as well as by hardware?

Fettkeewl, 2017-04-25:
As I understand it with your setup you can only awaken the device via external event by opening reed switch, from deep sleep that is?

chaeplin, 2017-04-25:
check my test graph, it works. change R and C .

chaeplin, 2017-04-25:
as 16 is not tied to RST, wakeup only works with external trigger.

Marmachine, 2017-05-14:
I guess this is exectly what i need. I'm using a WeMosD1 with DHT22 (temperature and humidity) sensor. 
The ESP reports the readings, then goes into deepsleep for 5 minutes.
To have the device wake-up, i had to put a 470 Ohm resistor between GPIO16 (D0) and RST (a 1K resistor didn't work!) Best solution would be schottky diode (cathode to gpio16, anode to rst), second best is resistor (470 Ohm should work, depending on used rst-pullup) worst solution is a direct connect (wire) -> you will have problems reflashing then!

Now I want to connect a door contact (breaks when door opens) and a pir (motion) sensor. So i am about to implement your setup, because i will need an external trigger. But combined with the above! Could you tell if this setup would work? I'm waiting for my parts to arrive before i can test this. Cheers!

Midgie75, 2017-05-19:
I combined chaeplin's monoflop with Marmachine's idea to connect GPIO16 with RST with a resistor. In my setup I had to use a 2K. Now it works! The NodeMCU I am using wakes up from DeepSleep via GPIO16-RTC-signal and via an external interrupt from my proximity sensor (Adafruit VCNL4010). Thanx guys!
If you like to take a look [follow my link.](https://github.com/Midgie75/ESP8266-Projects/wiki/%233-VCNL_NodeMCU_DeepSleep_GarageDoor_WatchDog)

closed 2021-07-15 by chaeplin