chaeplin/esp8266_and_arduino #42: What capacitor to use?

Marmachine, 2017-06-13:
Hi,
As i wrote earlier, i am very very happy with your project as it is exactly what i need, so again thanks for sharing! I'm using it with a WeMos D1 board and finaly all my parts have arrived so i've setup a test.

At first it wouldn't reset at all, then i figured i might have some issue with the capacitor i've used, i found out that:
when i shortcut the capacitor, then next after i open and close the reed contact, it does reset the board.

So, two questions: 
Do you think that using [this](http://www.ebay.com/itm/262026858649?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT) capacitor won't work. Which would you advice? I've used a 50V instead of a 16V, is that the trick? Can't really see them in your picture.

Just to be sure: The reed contact, is it closed when your (garage) door is closed? So with it all in rest (door closed) it connects to ground?

Thanks for your reply!
Marco

teemuhirsikangas, 2017-08-17:
In my setup, if using R5 as 2k resistor, it does startup but takes 1-10secs after is finally starts
if R5 is 470, 1k or something else, it does not wakeup. Don't have diode  yet, will try that once I got it

using 2k resistor is not reliable as sometimes it was way longer than 10sec

Fettkeewl, 2017-11-08:
Hi Marmachine, I have the same needs as you, to get chaeplins circuitry working with the ability to self-wake. Have you made any progress?

Marmachine, 2017-11-09:
Hi,

Unfortunately not. Actually i haven't been working on this project for some months since i ran into another issue.

It seems that the WiFi signal disturbs the PIR sensor which makes it go into a loop and call false alarms:

Motion detect > WiFi call > No Motion > Wifi call > PIR detects false motion > Wifi call > No Motion > Wifi call... etc.

I need to look into this, my first idea is to shield the sensor somehow, also i will test with a different (stronger) power adapter to make sure it's not a power related issue.

However, if you have any ideas, i will be glad to hear and ofcourse i'm willing to share my development as well.

Best regards,

Marco

Fettkeewl, 2017-11-11:
If its any help I vaguely remember reading that PIR's do get disturbed by wifi. Positioning them corretly..
If you have no need to constantly be connected then I suggest turning off your wifi and have it off untill motion trigger turns it on.

In regards to chaeplins circuit I might have cracked the nut that was bothering us.
It would seem that the resistor R1 and R5 have to high values for a proper negative pulse.

At the moment I managed to get it working with R1 as 100k and R4 as 1k. 
Currently measuring the current use. R1 = 300k not working, R1 = 100k working

closed 2021-07-15 by chaeplin