chaeplin/esp8266_and_arduino #43: Question - 48-door-alarm-deepsleep

SuoaJ, 2017-12-05:
I refer to your circuit here under the 48 door alarm project:

https://github.com/chaeplin/esp8266_and_arduino/blob/master/_48-door-alarm-deepsleep/pics/door_alarm_light_sleep_gpio_int.png

What pins do the "GPIO IN" and "LIGHT SLEEP IN" connect to on the ESP8266?

PProvost, 2018-10-19:
Per the code in that example, I think LIGHT SLEEP IN maps to pin 5, specified by this constant:

> #define DOOR_INT_PIN 5

I still haven't figured out where GPIO IN is supposed to go. There appears to be no reference to a 4th pin in the code.

@SuoaJ - Did you figure it out? I'm considering just not connecting it.

PProvost, 2018-10-19:
I think I found it here: https://github.com/esp8266/Arduino/issues/1488#issuecomment-248855617

| Circuit Label | Sketch Pin  |
|---------------|--------------|
| GPIO IN        | DOOR_PIN |
| LIGHT SLEEP IN | DOOR_INT_PIN |
| 16 | RST_PIN |
| RST | not in sketch, use RST on the board |

I would love for @chaeplin to confirm however.

closed 2021-07-15 by chaeplin